cooked/printhat #10: reflowed fuses got 5-15ohm resistance

cooked, 2018-07-05:
main bug, whereby the SMD fuses are soldered but the resulting series resistance is very high, causing shutdown/reset because of the voltage drop (under load e.g. when the steppers start)

- search for a better SMD fuse, with easier reflow specs

smark-, 2018-07-28:
I would keep the same fuses for the Breadboard but solder them by hand. They need to be replaced on the production BOM indeed

cooked, 2018-08-26:
looking at the specs, the wrong fuse had been selected. The former had an Itrip of 300mA, which is the same value at which the board was resetting during lab tests. 
The fuse is replaced with one of appropriate Itrip of 10A. Selected component is:
mfr part# 0501010.WR1
https://lcsc.com/product-detail/Others_Littelfuse_0501010-WR1_Littelfuse-0501010-WR1_C207003.html

closed 2018-08-26 by cooked