cooked/printhat #12: use 5V as TMC logic supply voltage

cooked, 2018-07-07:
currently, the internal 5V linear regulator is powered by Vmot which can be as high as 36V
this means poor efficiency especially at a high Vmot voltage and heating up of the components

consider using 5V from external DC/DC since it's readily available (check TMC specs for Vin min since the DC/DC output is now 4.9V )

smark-, 2018-07-31:
There are two ways, according to the datasheet:
- Support for the VCC Supply, where the external source only provides power to the digital logic
- Internal Regulator Bridged, where the external source is used to power both the digital logic and the analog part.
For the second option it is required to have a stable power supply with a ripple less than 5 mV. The switching power converter is not designed for low ripple, but for reduced components size and efficiency. I would therefore go for the first option, which already seems to improve the power dissipation significantly. This is the proposed circuit:
[image]
I would implement it right from the next iteration given the thermal issues found on the board. Vcc min is 4.6V

cooked, 2018-08-22:
I agree with the schematic suggested before and implemented below:
- 
[image]

closed 2018-08-26 by cooked